mist-devel/mist-board #159: Suggest BOSSA instead of SAM.BA or Sam_I_Am

dkerfoot, 2022-04-04:
Firmware update stalled. After 15 minutes I turned off and back on again - borked.

Not having any pre-Win 10 pcs anymore, I tried to get SAM.BA and Sam_I_Am working on a Ubunto on a stick and a RPI3. Very frustrating.

Found BOSSA by searching RasPi applications for SAM-BA. Installed the GUI and command line and ran and was flashed in a minute. Much, much easier

BOSSA Basic Open Source SAM-BA Application from Shumatech

gyurco, 2022-04-07:
Good to know an alternative solution exists. For Atmel's SAM-BA, the last usable is 2.1x, the 3.xx are horribly complicated for the simple task of unbricking.

harbaum, 2022-04-08:
Excellent find! Feel free to send a PR with  some instructions.

dkerfoot, 2022-04-08:
Thrilled to be paying back in a small way Till!

I'll test it on my Mac (worked great with both Win 10 and on my Pi3) and write up some directions in the next couple of days.

It really was super easy. Both Win10 and the Pi automatically recognised the MiST and selected the proper chip.

retrofun, 2023-01-03:
BOSSA works for me on my current debian/testing system.

I had to install the packages `libwxgtk3.0-gtk3-dev` and `libreadline-dev` to compile the code from https://github.com/shumatech/BOSSA

Then flashing MiST with the `firmware.bin` file via BOSSA GUI worked:

[image]

spicyjack, 2024-05-30:
I just used `bossa` from Homebrew on my M2 Macbook Air to unbrick my MiST box, so it definitely works on Apple Silicon Macs.

@retrofun's `bossa` command line commands above (`bossac -w firmware.bin`) in a Terminal window were all that was needed.

`$ brew info bossa
==> bossa: stable 1.9.1 (bottled), HEAD
Flash utility for Atmel SAM microcontrollers
https://github.com/shumatech/BOSSA
Installed
/opt/homebrew/Cellar/bossa/1.9.1 (7 files, 312.6KB) *
  Poured from bottle using the formulae.brew.sh API on 2024-05-29 at 19:03:56
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/b/bossa.rb
License: BSD-3-Clause`

EmmanuelKasper, 2024-05-30:
I have documented using Samba on Linux here: https://github.com/mist-devel/mist-board/wiki/Updating-the-IO-Controller-Via-Bossa-Under-Linux, will add a note that is works on Mac too via brew.

spicyjack, 2024-05-30:
I just added separate but related macOS instructions on that page...

https://github.com/mist-devel/mist-board/wiki/Updating-the-IO-Controller-Via-Bossa-Under-Linux-&-MacOS#installing-on-macos

EmmanuelKasper, 2024-08-09:
@harbaum we have the bossa method now documented in the wiki, feel free to review and close the issue.
 https://github.com/mist-devel/mist-board/wiki/HowToInstallTheFirmware. Worked great for me !

harbaum, 2024-08-10:
Cool. I currently don't have a working MiST, anymore. Gave the last one away as a replacement for a broken unit to one of Lothareks customers.

But I may re-activate one of my early prototypes one day.

closed 2024-08-10 by harbaum