mist-devel/mist-board #162: ini file and USB access

jotego, 2024-04-15:
These are questions I do not see answered in the [wiki](https://github.com/mist-devel/mist-board/wiki/DocIni). I would happily edit the answers if you reply here, or I will read them from there if you reply in the wiki:

- I see MiST can create a MISTCFG.INI file. How is this different from MIST.INI? Are there other .INI files?
- MiST also creates a `[CFG]` section in that file. How is that different from the `[mist]` section
- Does character case matters? is `[MIST]` the same as `[mist]`
- Having a MIST.INI file with just `USB_STORAGE=1` does not seem to take an effect on SiDi128. Is that expected?

Thank you

gyurco, 2024-04-15:
- MISTCFG.INI is created by the firmware itself when you save global settings. Currently only joystick mapping is stored in it, but it's possible in the future to have a configuration page in OSD.
- [CFG] is for saving the 'status' word (Save Settings in the system menu).
- Case doesn't matter as I see in https://github.com/mist-devel/mist-firmware/blob/master/ini_parser.c#L168
- I think I made usb_storage uncoditionally enabled in SiDi128. The reason is disabled by default in MiST that both serial port and usb storage cannot be enabled at the same time (hardware is out of possible endpoints).

jotego, 2024-04-16:
Thank you. I have edited the wiki to add this information.

closed 2024-04-16 by jotego