mist-devel/mist-board #164: Amstrad CPC Core and Alcon 2020

micv2, 2024-05-04:
Hi, 
Is there a way to load the game?

https://[abalore.com/alcon2020.html](https://abalore.com/alcon2020.html)

Best regards,
Paul

gyurco, 2024-05-04:
https://www.atari-forum.com/viewtopic.php?p=409608#p409608

micv2, 2024-05-04:
You are incredible, thank you so much 👍🙂

micv2, 2024-05-04:
Unfortunately the plane (sprite) disappears when I die.. It's probably a bug..

gyurco, 2024-05-05:
There's no sprite HW in the CPC, why just the ship disappers?
It works for me, BTW.

gyurco, 2024-05-05:
CRTC0 is better, with CRTC1, the game has some garbage during shooting.

micv2, 2024-05-05:
Hi, 
I also switched to CRTC0. Now it works better. I don't know what happened last night. Technically awesome game 👍

micv2, 2024-05-05:
Put 5 lives and 5 continuous and easy levels in the service menu. Then play. If I'm shot down, my plane will be invisible.

[image]