mist-devel/mist-board #165: Next core and MD- joy support ?

micv2, 2024-08-27:
Hello,

I want to set the Next core to MD pad or joystick. When I go to Settings in the NMI menu, I get an error loading enMf5.sys . 
Does that mean that a 3-button MD pad is not supported on MiST?

Regards,
MiC

[image]

gyurco, 2024-08-28:
I have no idea, the top-level sends all buttons to the ZXNext core:
```
      i_JOY_LEFT           : in std_logic_vector(11 downto 0);    -- active high  MODE X Z Y START A C B U D L R
      i_JOY_RIGHT          : in std_logic_vector(11 downto 0);    -- active high  MODE X Z Y START A C B U D L R
```
(Well, seems it doesn't send the last bit, MODE).

I don't know why "Error loading", maybe you should ask NextOS devs.

gyurco, 2024-08-28:
Probably "what the error loading of enmf5.sys means?"

micv2, 2024-08-28:
The last picture is from the real Next computer. The error occurs on the MiST when I want to go into the joystick menu. I can't actually select anything in the NMI menu, different errors appear. If I select "About" in the NMI menu, an error also occurs when loading and no information page is displayed. It's a shame, the "setup" doesn't work 100%. I have a real MD pad and can't use it on the MiST. It works on the real Next and I have three fire buttons on MD-pad.

Maybe I expect too much from MiST, can MiST technically do 3 fire buttons, or only via USB?

gyurco, 2024-08-28:
I don't see the connection with the 'about' error and the joystick, there must be something else.
On MiST, DB9 MD pads are not usable anyway (select pin is not physically available), only USB ones.

I just tried it, I can go into Settings, About, etc. in the NMI menu. There must be something with your NextOS.

micv2, 2024-08-28:
Interesting, which version of NextOS are you using?

micv2, 2024-08-28:
I have that version too. I can't save the time and date either. 
Regardless, the games work 👍

Thanks gyurco

gyurco, 2024-08-28:
MiST can supply the time and date from the USB RTC module. Maybe the crash because of the '0000' time (I have RTC).

micv2, 2024-08-28:
try without RTC?
If that is the case, then I will immediately buy RTC module 😉

gyurco, 2024-08-28:
I actually tried on SiDi128, which is basically the same core. Now I tried it on MiST (same SD Card), and I get 'error loading'. Maybe because it's slower and the loading timeouts.

gyurco, 2024-08-28:
Strangely it works with an older NextOS. Maybe they reduced the timeouts?

micv2, 2024-08-28:
I have two MiST v1.3 and v1.5. The same error on both. Both have the latest firmware.

I once watched the next demo "ScrollNutter" and spontaneously pressed F9->About. It loaded, and the settings joystick also loaded. Then MiST turned off, started everything again, pressed F9 -> settings, and error again.

Strange error, maybe timing or RTC?
I have tried old NextOS versions, down to v1.3.2

gyurco, 2024-08-28:
I don't think it's the RTC. Probably SD Card timeouts.

micv2, 2024-08-28:
Maybe you are on the right track.

Small gift, I bought 👍😉

[foryou.zip](https://github.com/user-attachments/files/16789160/foryou.zip)

micv2, 2024-08-28:
I have found a solution for my Aliens Neoplasma and my Joycard from larek with MD extension. 
With the MD extension it doesn't work with the MiST.

https://youtu.be/c5xLDLd1rd4?si=iL_ArHoALacynq4q&t=1

But if I use the USB- PC extension, then I can play Aliens Neoplasma with all functions without configuration! Great!
larek speaks in Polish, but the videos are self-explanatory.

https://youtu.be/pPEP-rNtKBc?si=uEnp4o1jqBzpwSgQ&t=1

gyurco, 2024-08-28:
Cool.
On SiDi128, it works with the MD setting :)
As it's possible to enter the settings menu...

micv2, 2024-08-28:
SiDi128 is powerful 😎👍

HiroDevelop, 2024-08-29:
Hey, can't we make this machine more cheaper? lol

gyurco, 2024-08-29:
You can try, schematics are available.

gyurco, 2025-02-16:
It works with the latest Next OS distribution.
Closing this, as it's not about issues, but general support questions, they're better to ask on the atari-forum.

closed 2025-02-16 by gyurco